Optic00/ha-opti-akkusteuerung #5: Opti Automatik

fisch55, 2024-12-14:
Hi, so ich habe mal wieder eine Frage....

Ich habe jetzt die optiAutomatik als Automatisation aus deiner Vorlage erstellt.
Musste jedoch noch einen input_boolean.hausakku_aus_netz_laden anlegen.
Jetzt meine Frage...wenn ich die OptiAutomatik nutzen möchte, was genau muss ich bei der AkkuSteuerung einstellen. 
Akku schnell laden?
Akku automatisch ? ..

Wäre super wenn du mir hier noch etwas licht ins dunkle bringen könntest.

VG

Skybarks, 2024-12-14:
Akku automatisch ist der " Normalzustand". Also wenn die SMA Steuerung läuft, ist das Akku automatisch. Wenn irgend etwas nicht klappt macht es auch Sinn, wieder darauf zurück zu stellen.
Wenn Du die Opti-Automatik an hast (und nur dann), und ein Trigger löst aus, wird automatisch umgeschaltet. 
Heißt z.B. es kommt viel Leistung vom Dach, und dein Akku steht bei über 50% dann lädt er mit der voreingestellten C Ladung (z.B. 0.2).  Wird der voraussichtliche Ertrag aber kleiner als das was du eingestellt hast, lädt er wieder...
Die Trigger kannst Du ganz nach Bedarf einstellen, je nach deinem Nutzungs-Szenario.
Falls Du ein paar use cases brauchst schick mir ne PM, dann melde ich mich nächste Woche und wir können das gerne durcharbeiten ...

fisch55, 2024-12-15:
Super , ich danke dir vielmals!!! Tibber habe ich nicht…. bekomme ich noch einige Entitäten angezeigt, die ich wohl für opti benötige…. oder nicht ?
Grüße und DANKE !!!
Eine Entität habe ich bereits angelegt , jedoch keine Ahnung für was diese ist. 
input_boolean.hausakku_aus_netz_laden 

[image]

Skybarks, 2024-12-17:
Wenn Du Tibber nicht nutzt, kannst Du die Trigger, in denen diese Entitäten verwendet werden einfach deaktivieren oder tatsächlich löschen (aber wer weiß - vielleicht brauchst Du sie ja doch irgendwann :-) )
Habe ich auch so gemacht.

Vielleicht nochmal kurz wie ich es eingerichtet habe, und ich bin echt super happy damit:

1. Die "Akkusteuerung" Automatik aus dem Git kopieren.
2. die Helper und Schalter anlegen die Du brauchst (stehen in der Anleitung. Achtung: Die Namen müssen stimmen! Wenn Du mit Copy & Paste arbeitest, kannst Du z.B. auf die Nase fallen wenn Du "input_number.akkusteuerung_ladestaerke_soll" kopierst, weil input.number schon drin steht, sobald Du den Wert als numerische Eingabe angibst.) 
4. Testen. Was jetzt funktionieren sollte ist ein manuelles umschalten zwischen Laden, Pause und Entladen. Um alles wieder auf Standard zu  stellen, einfach wieder "Akku Automatisch" auswählen, dann übernimmt die SMA Steuerung.

5. Jetzt wirds spannend: Den Opti Automatik Teil kopieren (bitte erstmal deaktivieren, und auch den Schalter "opti automatik" den Du vorher angelegt hast aus lassen.)
6.  Anpassen. Heißt Deine(n) WR eintragen, die nicht benötigten Tibber Teile deaktivieren - das geht sogar ganz gut in der graphischen Ansicht. Eventuell nochmal die Wunschwerte prüfen, also ab wann soll er volladen, wie hoch soll die 0.2c Ladung sein etc.
7. Dann aktivieren und erstmal ins Log schauen.
8. Sind nach 1-2 Minuten keine Fehler im Log, die "opti-automatik" einschalten und wieder ins Log gucken.
9. Sich freuen weil es klappt, oder hier kurz Bescheid geben :-)