Optic00/ha-opti-akkusteuerung #6: SMA SBS 3.7 mit BYD HV-Box und SMA STP 6.0 SE mi BYD HVM

Michl09, 2025-01-06:
Hallo,
die beiden Batterien im Titel laden bzw. entladen sich gegenseitig.
Wäre diese yaml eine Vorlage um das zu vermeiden? 
Ich bin kein Programmierer, könnte es auch als HACS gemacht werden?
Viele Grüße
Michl

Skybarks, 2025-01-08:
Hallo Michl,
das sollte mit dieser Steuerung problemlos klappen.
Du kannst ja z.B. eine Automatik anlegen, die erst Batterie Nummer 2 lädt wenn Batterie 1 voll ist, oder das Ganze sogar per verfügbarer PV Energie aufteilen.
Als Hacs wäre das Ganze zwar möglich, aber aufgrund der 1000 Möglichkeit schwer zu supporten.
Darum ist die Steuerung eher als Werkzeugkasten zu verstehen.
Sobald man die Basis mal am laufen hat, ist der Rest ein Kinderspiel und lässt einen immer wieder neue Möglichkeiten entdecken.

Michl09, 2025-01-08:
> Chris

Hallo Chris,
danke für die Antwort. Momentan kann ich mit SMA Devices plus und SMA Solar
jeweils einen Wechselrichter problemlos auslesen.
Ich habe noch keine yaml direkt in HA hinzugefügt. Was denkst du welche der
3 yaml würde ich für meinen Fall benötigen? Wo müsste ich sie hinkopieren
in HA?
Gruß Michl

closed 2025-01-12 by Michl09

Skybarks, 2025-01-12:
Ich würde wie folgt vorgehen: 
Zu allererst ein Full Backup machen :-)
Dann die Akkusteuerung für den ersten Hybrid WR einbauen. Ohne Opti-Automatik.
Den WR in der config.yaml dann evtl. als WR1 benennen.
Dann laut Anleitung die Schalter und Zahlenwerte eintragen und testen. Darauf achten, dass sich alles auf den WR1 bezieht.
Jetzt solltest Du schon mit einer einfachen Logik einen WR bzw. die angeschlossene Batterie manuell steuern können.
Sobald Du das hast, dasselbe mit dem WR2. Die Schalter und Werte müssen dann natürlich entsprechend umbenannt werden, damit die Schalter auch den WR2 steuern.
Wieder testen, und wenn es klappt eine einfache Automatik bauen, die dem entspricht was Du Dir vorstellst (Tageszeit/PV Leistung/Status der Batterien wie Laden/Entladen/Pause).
Wir können das auch gerne Ende nächster Woche zusammen mal durcharbeiten...
Schöne Grüße
Chris

Michl09, 2025-01-15:
Hallo Chris,
ich würde mich freuen, wenn wir das zusammen durcharbeiten könnten. Ich
gebe dir mal meine email Adresse. Sie lautet: ***@***.***
Skype hätte ich auch.
Gruß Michael

Skybarks, 2025-01-20:
in der /homeassistant/configuration.yaml müsste es ganz unten so aussehen.
Achtung! Die IP Adressen hinter Host durch die Adresse Deiner Hybrid WR ersetzen!

modbus:
  - name: sma-sr_wr1
    type: tcp
    host: 192.168.xx.xxx
    port: 502
    delay: 2
    message_wait_milliseconds: 60
    timeout: 5
    sensors:     
      - name: SMA-STP-SE_Temperatur
        data_type: uint32
        device_class: "temperature"
        unit_of_measurement: "°C"
        slave: 3
        address: 30953
        #count: 2
        scale: 0.1
  - name: sma-sr_wr2
    type: tcp
    host: 192.168.xx.xxx
    port: 502
    delay: 2
    message_wait_milliseconds: 60
    timeout: 5
    sensors:     
      - name: SMA-STP-SE_Temperatur
        data_type: uint32
        device_class: "temperature"
        unit_of_measurement: "°C"
        slave: 3
        address: 30953
        #count: 2
        scale: 0.1
  

Jetzt solltest Du in der Automation "Akkusteuerung" nur die Entitäten auf sma-sr_wr1 ändern.
Dann die Automation kopieren, umbenennen auf z.B. Akkusteuerung 2 und darin die Entitäten auf sma-sr_wr2 ändern.
Am einfachsten geht das Ändern wenn Du auf "in YAML bearbeiten" klickst, und einfach nach sma-sr_wr suchst / ersetzt.

Dan noch die Werte laut Beschreibung setzen, eventuelll auch pro WR, falls Du unterschiedliche Werte möchtest, und testen ob es klappt. Also einfach mal auf "Akku Pause", und im SMA Portal schauen, ob der WR das korrekt steuert. Dann zurück auf "Akku automatisch" , und er sollte wieder ins Standard Verhalten zurückgehen.

Wenn das klappt, machen wir weiter :-)

Skybarks, 2025-01-20:
[image]

die Einrückungen in der configuration.yaml sehen bei mir übrigens so aus.

Michl09, 2025-01-20:
Dankeschön!
Ich versuche es. Weiß nicht recht wann es schaffe :-)
Bei Skype wäre ich unter Michael F zu finden. Ich trage eine Brille und
einen Fahrradhelm. Ich würde mich freuen, wenn du dich meldest :-)

Michl09, 2025-02-09:
Hallo Skybarks,
die Config und die Automation konnte ich einbinden.
Als erstes würde es mir reichen wenn der WR2 nur lädt wenn WR1 lädt. Oder
WR2 nur entlädt wenn WR1 entlädt.
Ich denke die Helfer für Überschuss brauche ich nicht.
Wie bekomme ich den Helfer zur Auswahl des Akku Modus rein?
Und wie müsste ich dann weitermachen.
Gruß Michl09

Skybarks, 2025-02-09:
Hi Michel,
habe eben bei Skype geschaut - die Kombi Fahrradhelm & Brille aber nicht unter dem Namen gefunden :-)

Genau: Den Überschuss Helfer lassen wir mal weg.
Den Helfer zur Steuerung bekommst Du so rein:
Einstellungen/Geräte&Dienste
Dann Helfer anklicken
Rechts unten "Helfer erstellen"
Dann nimmst Du am besten das "Dropdown Menü" und hinterlegst als Optionen:
Akku Automatisch
Akku schnell Laden
Akku schnell Entladen
Akku Pause
Akku 0.2C Laden
Die Input Numbers auch unter den Helfern anlegen:
input_number.akkusteuerung_ladestaerke_soll 
input_number.akkusteuerung_entladestaerke_soll
input_number.akkusteuerung_02c_ladestaerke
Aber Achtung: "input_number" hängt der Home Assistant schon alleine rein. Du gibst also nach Auswahl der "Zahelnwert Eingabe" als Name nur noch akkusteuerung_ladestaerke_soll ein.
Ob Du das als Schieberegler oder Zahleneingabe machst, ist Gemackssache :-)

Skybarks, 2025-02-09:
OK. Dann fehlt jetzt die Anweisung für den WR, dass er in den Pause Modus gehen soll.l

Schau bitte ob Du in deinen Automationen die  Akkusteuerung findest, und ob sie auch aktiviert ist
In der Akkusteuerung muss die IP deines WR richtig eingetragen sein.

[image]

CarlosEllan, 2025-02-09:
Dummerweise geben die Dokumente von SMA nicht für jedes Gerät her, ob die beiden Register unterstützt werden. So gibt es bei meinem Sunny Island die Register gar nicht (laut Beschreibung). Zum Glück funktioniert aber alles. Nur kann man halt bei WO-Register nicht prüfen, ob die Automation die Register hat beschreiben können.
Andererseits sind die Scripte ja nicht kompliziert.
Ich würde prüfen, ob die IP-Adresse des Batteriewechselrichter stimmt. Ich hab auch nur die Grundfunktionalität und die Helfer eingerichtet.

Skybarks, 2025-02-09:
> Ich hab auch nur die Grundfunktionalität und die Helfer eingerichtet.

Im Prinzip ist genau das auch der Kern der Idee, würde ich sagen (ist aber nicht meine Integration :-) ).

Wenn die Basis klappt, hat man einen tollen Werkzeugkasten, den man nach Lust und Laune einsetzen kann. 

Ich entlade zwischenzeitlich zum Beispiel morgens und abends meinen Akku ins Netz, wenn ich weiß dass über den (nächsten) Tag genug Energie kommt, ich genug Reserve habe, und ich somit das Netz stützen kann. 

Also von der Eigenoptimierung bis zur ökologisch orientierten Ausrichtung kann man echt alles damit basteln...

Michl09, 2025-02-11:
Es funktioniert noch nicht. In meiner configuration.yaml müsste der Name
passen.
modbus:
- name: sma-sr_wr1
type: tcp
host: 192.168.178.46
port: 502
delay: 2
message_wait_milliseconds: 60
timeout: 5
sensors:
# - platform: modbus
# registers:
- name: SMA-STP-SE1_Temperatur
data_type: uint32
device_class: "temperature"
unit_of_measurement: "°C"
slave: 3
address: 30953
#count: 2
scale: 0.1
- name: sma-sr_wr2
type: tcp
host: 192.168.178.130
port: 502
delay: 2
message_wait_milliseconds: 60
timeout: 5
sensors:
# - platform: modbus
# registers:
- name: SMA-STP-SE1_Temperatur
data_type: uint32
device_class: "temperature"
unit_of_measurement: "°C"
slave: 3
address: 30953
#count: 2
scale: 0.1

scale: 0.1

In der Automation für WR1 habe ich bei hub überall angepasst:
 hub: sma-sr_wr1

In der Automation fällt mir dieser Fehler auf: e ist undefinded
Wie könnte ich den beheben?

[image: grafik.png]

Wie kann ich den Schalter und das Input Select Menü eindeutig machen, so
dass er nur für den WR1 zuständig ist?

Übrigens: mein Skype Name ist live:mi.finkenzeller

Wenn Du tatsächlich mal drüber schauen könntest, würde ich mich sehr freuen
:-)

Michl09, 2025-02-22:
@Skybarks Konntest du meine Skype Nachricht lesen. Würde es bei dir morgen gehen?

Skybarks, 2025-02-22:
Hi Michl, 
an Wochenenden ist es bei mir immer schwierig, besonders Abends.
Geht bei dir mal unter der Woche? Am besten Vormittags? 
LG

Michl09, 2025-02-23:
Hi Chris, 
am Dienstag vormittag würde es gehen. Wir früh würdest du beginnen können?
Gruß
Michl

Skybarks, 2025-02-23:
Dienstag ist gut. 9.00 Uhr?

Michl09, 2025-02-23:
Ja, passt. Dann bis dahin bei Skype :-)