Optic00/ha-opti-akkusteuerung #7: Flash-Speicher und 'zyklisches' Schreiben

CarlosEllan, 2025-01-13:
So, hab jetzt auch mal die ersten Automationen und Helfer angelegt.
Funktioniert soweit auch mit einem Sunny Island 8.0.
Bei der Automation 'Akkusteuerung' sind mir noch 2 Tippfehler bei den Alias aufgefallen, die aber keine Auswirkung haben: 'Akku schnell Entaden' und der Alias 'Wenn Akku Automatik' gehört wohl vor Zeile 107.
Aber zum Issue: Gibt es inzwischen eine offizielle Aussage von SMA, ob die Register 40149 und 40151 problemlos auch mehrfach am Tag beschrieben werden können? Die Modbus Dokumentation ist da sehr vage und je nach genutztem Flash-Speicher kann das kritisch werden.
Auf jeden Fall schon mal Danke für die Scripte. Sobald ich sie für meine Hardware und Bedürfnisse angepasst habe, melde ich mich.

Skybarks, 2025-01-13:
Es gibt für die SMA Produkte im jeweiligen Download - Bereich des Produkts Listen mit den entspr. Modbus Parametern:
https://www.sma.de/service/downloads
Beim meinen beiden STP 10 SE und STP 10 z.B. sind die Register 40149 und 40151 als WO Register genannt. 
Für den Sunny Island 8 gibt´s das sicher auch.
Probleme gibt es laut SMA nur beim zyklischen Beschreiben von RW Parametern (Quelle: Modbus Beschreibung SMA)
Ich gehe mal davon aus, dass ein WO Register also beliebig oft geschrieben werden kann, da der Vorgang nur in den Arbeitsspeicher, und nicht auf die Flash Karte schreiben sollte.
Dazu gabs im Photovoltaik-Forum auch mal ne Frage (#57):
https://www.photovoltaikforum.com/thread/206718-sma-stp10-0-3se-40-welcher-modbus-register-zum-laden-der-batterie/?pageNo=6

CarlosEllan, 2025-01-13:
Ich versuche ja auch schon länger, an die Registerbeschreibung der SI 8.0H-11 zu kommen. Leider ist das im Archiv von SMA sehr dürftig beschrieben, und in dem Dokument kommen die 40149 und 40151 nicht vor. Nur in einer allgemeinen Modbus Doku, die im Iobroker-Forum verlinkt wurde, konnte ich diese finden.
Da die beiden Register auch beim SI 8.0H-11 read-only sind (zumindest kommt nicht der vorher reingeschriebene Wert zurück), gehe ich davon aus, daß diese nicht im Flash gehalten werden - also unkritisch.
Ich schau mir jetzt mal die opti-automatic Automation an, hab aber etwas andere Vorstellung, wie ich meinen Speicher mit Tibber optimieren werde. Hier stehen zwar 40kWh LFP rum, aber die neue WP zieht auch über 1,5kW fast durchgängig. Ab März langweilt sich der Speicher eh mehr, als daß er belastet wird. Aber mit Modul 3 nach §14A dann schon interessanter - angeblich arbeitet SMA ja an einer Lösung mit Lichtblick. Allerdings funktioniert schon das prognosebasierte Solarladen bei SMA eher unzuverlässig.

Skybarks, 2025-01-13:
> Allerdings funktioniert schon das prognosebasierte Solarladen bei SMA eher unzuverlässig.

Oh ja, die Erfahrung habe ich auch gemacht :-)
Was bei mir jetzt gut klappt, ist eine Kombination aus der Optic00 Akkusteuerung, der SMA-SEMP Anbindung von littleyoda (zu finden hier: https://github.com/littleyoda/ha-sma-semp), und ein paar HA-Automatiken die ich darauf aufgesetzt habe. 
Seitdem regelt der HA meine beiden WR und die Batterie so exakt, dass ich manchmal selbst staune.

CarlosEllan, 2025-01-20:
So, bin inzwischen auch so weit gekommen, das das script 'Akkusteuerung' funktioniert.
Hab allerdings in Zusammenhang mit evcc ein 'Feature' festgestellt:

Wenn man in evcc unter Konfiguration/Hausbatterie den Schalter

` Verhindere Entladung im Schnell-Modus und bei geplantem Laden.`

eingeschaltet hat, funktioniert 'Akku schnell Laden' + Fahrzeug laden über evcc nicht. Unabhängig von der Einstellung, wie das BEV geladen werden soll.
Wenn der Schalter eingeschaltet ist, und unter Akkusteuerung 'Akku schnell Laden' eingestellt ist oder man dies über eine Automation umstellt, wird die Hausakku-Ladung unterbrochen, sobald evvc eine Ladung des Auto startet (egal , ob automatisiert oder wenn man es manuell auswählt). Reihenfolge ist dabei egal. Ich denke, daß ivcc hier auch auf den Akku zugreift und sich das dann beißt. Ist für mich unkritisch, aber evtl. interessant, falls auch jemand beide Integrationen nutzt.

alphabravocharly72, 2025-02-18:
@CarlosEllan 
Genau sowas suche ich schon länger. 
Meine Config ist Core1, 3x SI 4.4-12, 20kWh Bat, Data Manager M.
Möchte es auf mein System umschreiben. Wäre es möglich das du mir deine yaml zukommen lässt? 
Danke

CarlosEllan, 2025-02-19:
Das ist ja nicht in nur einer yaml.
Ich nutze ja nur die sma-se-akku-steuerung.yaml mit den drei Modi Akku schnell Laden, Akku Pause und Akku Automatisch. Akku schnell Entladen funktioniert bei mir auch, hab ich aber noch keinen Einsatz für (evtl. später, wenn meine Einspeisung über Direktvermarktung laufen sollte).
Hier meine sma-se-akku-steuerung.yaml:

```
alias: Akkusteuerung
description: ""
triggers:
  - entity_id:
      - input_select.akkusteuerung_sma_wr
      - input_number.akkusteuerung_ladestaerke_soll
      - input_number.akkusteuerung_entladestaerke_soll
    from: null
    to: null
    for:
      hours: 0
      minutes: 0
      seconds: 1
    trigger: state
  - minutes: /30
    enabled: true
    trigger: time_pattern
  - event: start
    trigger: homeassistant
conditions: []
actions:
  - alias: Wenn Akku schnell Laden ...
    if:
      - condition: state
        entity_id: input_select.akkusteuerung_sma_wr
        state: Akku schnell Laden
    then:
      - data:
          hub: sma-sr_wr
          address: 40151
          slave: 3
          value:
            - 0
            - 802
        action: modbus.write_register
      - delay:
          hours: 0
          minutes: 0
          seconds: 5
          milliseconds: 0
      - data:
          hub: sma-sr_wr
          address: 40149
          slave: 3
          value:
            - 65535
            - >-
              {{ 65535 - states('input_number.akkusteuerung_ladestaerke_soll') |
              int * 0.3333 }}
        action: modbus.write_register
  - alias: Wenn Akku schnell Entladen ...
    if:
      - condition: state
        entity_id: input_select.akkusteuerung_sma_wr
        state: Akku schnell Entladen
    then:
      - data:
          hub: sma-sr_wr
          address: 40151
          slave: 3
          value:
            - 0
            - 802
        action: modbus.write_register
      - delay:
[... 350 lines cut ...]
    address: 30777
    data_type: uint32
  - name: SMA Power L2 A 
    unique_id: SMAPowerL2A
    device_class: power
    unit_of_measurement: W
    slave: 3
    address: 30779
    data_type: uint32
  - name: SMA Power L3 A 
    unique_id: SMAPowerL3A
    device_class: power
    unit_of_measurement: W
    slave: 3
    address: 30781
    data_type: uint32
  - name: SMA-STP_Temperatur"
    unique_id: StpTemp
    data_type: int32 #S32
    device_class: "temperature"
    unit_of_measurement: "°C"
    slave: 3
    address: 30953
    scale: 0.1
```

alphabravocharly72, 2025-02-20:
Danke für den code.
Hätte ein paar Fragen dazu
Wo und wie hast du diese inputs her?
- entity_id:
      - input_select.akkusteuerung_sma_wr
      - input_number.akkusteuerung_ladestaerke_soll
      - input_number.akkusteuerung_entladestaerke_soll
      - input_number.akkusteuerung_02c_ladestaerke

entity: input_boolean.akku_opti_automatik
  - type: tile
    entity: input_boolean.akku_nach_preis_laden
  - type: tile
    entity: input_boolean.pv_module_nicht_verfugbar

entities:
      - entity: input_number.akkusteuerung_ab_welchem_restertrag_vollladen
        name: Restladeschw.
     - entity: input_number.akkusteuerung_wr_ac_ueberschuss_grenze
        name: WR AC-Grenze
     

Danke für deine Hilfe
alpha

CarlosEllan, 2025-02-20:
Steht oben in der Readme. Das sind alles Helfer, die sich in Home Assistant in der UI über Geräte & Einstellungen unter Helfer einrichten lassen. Am besten mal in Youtube nach Hilfsvideos über Helfer in HA suchen.

alphabravocharly72, 2025-02-20:
wenn ich eine Zahlenwert-Eingabe erstellen will, wird es nicht übernommen nach dem speichern.
Name: akkusteuerung_entladestaerke_soll
Minimalwert: 100
Maximalwert: 10000
Anzeigemodus: Schieberegler
Schrittweite und Masseinheit ist 1 und leer.

CarlosEllan, 2025-02-21:
Also: in die config gehört nur der include für die modbus.yaml.

Die sma-se-akku-steuerung.yaml ist eine Automation, die wie die Helfer über die Geräteeinstellung eingerichtet wird. Am einfachsten über die Einstellungen eine neue Automation anlegen und den Code der sma-se-akku-steuerung.yaml dort reinkopieren und die Automation sinnvoll benennen.
Über die Helfer und eine entsprechende Karte wird dann der Status jeweils gesetzt.
Besser wirklich mal YouTube Videos schauen, wo das Konzept und die Grundlagen von Automationen und Helfern gezeigt wird.
Grundvoraussetzung ist auch, daß man über den Modbus auf den Sunny Island zugreifen kann.

closed 2025-02-25 by CarlosEllan