Optic00/ha-opti-akkusteuerung #9: Funktion für SMA Sunny Boy Storage SBS 3.7 Batteriewechselrichter SBS3.7-10

stasihasi, 2025-04-08:
Hi,

ich würd gern dein Tool für einen Batteriewechselrichter von SMA (Sunny Boy Storage SBS 3.7) nutzen (Firmware 4.7.19.R). Dabei würd ich am liebsten die Modbus-Adressen nutzen. Jedoch scheinen diese andere zu sein. Firmwareversion des SunnyHomeManagers 2.0 ist 2.15.7.R.
Hast du eine Idee, wo ich die Doku hierzu finde? So könnte ich die Adressen in deinen Automationen einfach austauschen für meinen Wechselrichter.
Finde das Projekt super spannend und danke schon mal für deine Antwort!

Grüße

Optic00, 2026-09-05:
Korrektur zu meinem Juli-Kommentar: Ein Mapping allein reicht für SBS-Unterstützung nicht. Es normalisiert die Eingangssensoren für die Strategie. Zum Steuern braucht es zusätzlich einen für den SBS geprüften Hardware-Adapter mit passenden Registern, Datentypen, Schreibabläufen und nachgewiesener Wirkung der einzelnen Modi.

Die Architektur trennt beides inzwischen klar: [Modus-Contract](https://github.com/Optic00/ha-modbus-akku-adapter/blob/main/docs/modus-contract.md). Den STP-SE-Adapter können wir nicht allein wegen ähnlich benannter Register als SBS-kompatibel erklären. Das Issue bleibt offen, bis Mapping und Adapter anhand der passenden Gerätedokumentation und eines SBS-Tests bestätigt sind.

Optic00, 2026-09-15:
Die aktive Entwicklung ist inzwischen auf eine HACS-Integration umgestellt. Der SBS 3.7 ist dadurch weiterhin nicht als unterstützt bestätigt; das Issue bleibt offen.

Für einen passenden Geräteadapter brauchen wir die zum Modell und zur Firmware gehörende Modbus-Dokumentation sowie einen zunächst rein lesenden Test am SBS. Schreibregister, Betriebsarten und Rückfallverhalten müssen separat belegt werden. Die STP-SE-Anbindung sollte nicht versuchsweise auf dem SBS aktiviert werden.

Die frühere YAML-Lösung bleibt im Branch `legacy-yaml`. Neue SBS-Unterstützung würde in der HACS-Anbindung entstehen; ein Sensormapping allein reicht dafür weiterhin nicht.

Die weitere Einordnung steht in der Qualitäts-Roadmap #85.

Optic00, 2026-10-21:
Ich schließe das Issue als „derzeit nicht unterstützt“. Für den SBS 3.7 fehlen weiterhin die bestätigten Schreibregister und ein lesender Test am Gerät; ohne beides kann ich keinen Adapter bauen und den STP-SE-Adapter für dein Modell nicht freigeben.

Wenn du die Modbus-Dokumentation zu deinem Modell und deiner Firmware hast und einen lesenden Test machen willst, öffne dieses Issue wieder oder erstelle ein neues. Die unterstützten Geräte stehen in der README.

closed 2026-10-21 by Optic00