Optic00/ha-opti-akkusteuerung #10: wie installiere ich das in home assistant

sbugs19, 2025-07-06:


Skybarks, 2025-07-06:
Hi Sbugs,
das Gute (und gleichzeitig das schlechte) ist, dass Du gar Nichts installieren musst. 
Das Ganze besteht aus Automations und ein paar Einträgen in den yaml Dateien deiner Wahl, also z.B. configuration.yaml
Out of The Box ist da schwierig, da die Einsatzszenarien unendliche sind, genau wie die Anforderungen.
Vor einiger Zeit gab es zum Beispiel den Wunsch, zwischen zwei Hybrid WR zu balancen, damit sich die Batterien nicht gegenseitig laden. Oder eben einfach nur Laden nach Lust, Laune, Sonne oder Windrichtung :-) 
Alles machbar, aber einen Standard gibt's da leider nicht.

Am besten schilderst Du mal, was genau Du realisieren möchtest, dann können wir Dir gerne helfen...
VG
Chris

Skybarks, 2025-07-06:
Ah. Dafür passt das ja  perfekt. Genau sowas hat Ben auch gebaut, soweit ich weiß.
@Optic00 :Ben,  wie sieht's bei Dir aus? Tibber, dynamische Strom-Tarife etc. sind ja Deine  Domäne. Da bin ich bisher ziemlich unbeleckt :-)